raspberrypi/documentation #3866: Broken links in ai-camera / details.adoc

jfjoly, 2024-10-02:
The links in the [rpicam-apps](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/accessories/ai-camera/details.adoc#rpicam-apps) paragraph are broken. Both give a 404 error. 
- https://github.com/raspberrypi/rpicam-apps/blob/post_processing_stages/imx500_post_processing_stage.hpp
- https://github.com/raspberrypi/rpicam-apps/blob/post_processing_stages/imx500_mobilenet_ssd.cpp

https://github.com/raspberrypi/documentation/blob/5c1691cca599b22f53d383bc8ca887e2b8286770/documentation/asciidoc/accessories/ai-camera/details.adoc?plain=1#L86

lurch, 2024-10-02:
`imx500_post_processing_stage.hpp` seems to be at https://github.com/raspberrypi/rpicam-apps/blob/main/post_processing_stages/imx500/imx500_post_processing_stage.hpp but I dunno where the other file should be.
ping @naushir

naushir, 2024-10-03:
This should be fixed with https://github.com/raspberrypi/documentation/pull/3867.

closed 2024-10-03 by nathan-contino