raspberrypi/documentation #3880: Conflicting documentation - debugprobe on Pico 2

arc12, 2024-10-09:
A This page - https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#debugging-using-another-pico-series-device - states:
"Pico 2 is not yet supported by debuprobe_on_pico debugging software. "

B The getting started guide states :
"One Pico or Pico 2 can reprogram and debug another using the debugprobe firmware, which transforms the Pico or Pico 2
into a USB → SWD and UART bridge."

C. The README for this repo describes how to build debugprobe_on_pico for Pico 2.

Is "A" in need of revision?

nathan-contino, 2024-11-07:
Fixed!

closed 2024-11-07 by nathan-contino

Palantir555, 2024-11-10:
Does it take a while for the changes to be applied to the docs site? I'm still seeing the warning message on the website, which had me double-checking for a while. No biggie regardless, just letting you know in case it wasn't submitted.

lurch, 2024-11-10:
> Does it take a while for the changes to be applied to the docs site?

Yes. The documentation has already been fixed on the [`develop` branch](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/microcontrollers/pico-series/utilities.adoc) but it won't be visible at https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#debugging-using-another-pico-series-device until @nathan-contino merges the `develop` branch into the [`master` branch](https://github.com/raspberrypi/documentation/blob/master/documentation/asciidoc/microcontrollers/pico-series/utilities.adoc) :slightly_smiling_face: