raspberrypi/documentation #3885: Accessories - Camera -Hardware Specification Table.  Maximum exposure time for V1 camera is out of date.

sandyol55, 2024-10-15:
Recent updates to the kernel driver for the V1 camera now allow a max exposure time of  ~ 3.28 secs.
Hardware specification table currently shows a Maximum exposure time (seconds) of 0.97.

lurch, 2024-10-23:
ping @naushir @davidplowman and @6by9

6by9, 2024-10-23:
If you hold fire I may get it up to the same 6 seconds as the firmware driver.

It'll need a dtoverlay override to enable it though as it'll select a lower link frequency globally (I really don't fancy making link frequency user controllable).

nathan-contino, 2024-11-07:
Fixed. @6by9 , when that 6 second time lands (enabled by dtoverlay), feel free to post it on this thread or just send it my way and I'll add a tiny little instruction on it in the relevant doc.

Thanks all!

closed 2024-11-07 by nathan-contino