raspberrypi/documentation #3889: Tutorial: how to use a raspberry pi in kiosk mode - why desktop??

qbicdesign, 2024-10-21:
[https://www.raspberrypi.com/tutorials/how-to-use-a-raspberry-pi-in-kiosk-mode/](https://www.raspberrypi.com/tutorials/how-to-use-a-raspberry-pi-in-kiosk-mode/)

I don't understand why the tutorial is based on Pi OS with desktop if it's only displaying a kiosk browser.
The desktop isn't used.
On older Pi's especially, keeping bloat to a minimum is important.
Or am I missing something?

nathan-contino, 2024-10-21:
Thanks for the feedback and the question. Reading through the tutorial again myself, I see why you might wonder!

@JamesH65 is correct -- in order to show you a GUI on screen, Chromium and Firefox require you to run a desktop. Honestly, the desktop does all the hard work of rendering stuff to the screen, once you've got that kiosk mode is just a way to render a browser page within that framework. As far as I know, kiosk mode without a desktop would be a pretty massive amount of work, as you'd have to provide most of the pieces of a desktop through the Lite OS. Stuff like Wayland or X would make a Lite OS with GUI-rendering capabilities almost as heavy as full fat Raspberry Pi OS. [disclaimer: if someone more knowledgeable than myself about GUI-fu knows differently, I am happy to reconsider this stance]

I'll add some text elaborating on this point to make the need for GUI rendering more obvious.

closed 2024-10-21 by nathan-contino

qbicdesign, 2024-10-21:
I used to use the following method in Raspbian Stretch, for which you don't need a desktop, you only needed X, Openbox and the browser:
https://die-antwort.eu/techblog/2017-12-setup-raspberry-pi-for-kiosk-mode/
It was very simple and quick to get up and running. 
I think this method no longer works in the latest Pi OS, as it's structure has changed a lot (but I'm gonna try),  and there seems to be issues now with both Firefox and Chromium running on 32bit (older Raspberry Pis).
I got something similar up and running without any desktop on a Pi 4 using Chromium, but it didn't work on a Pi model B+ which i have about 4 of, and I don't want to replace if I can avoid it...

qbicdesign, 2024-10-22:
Update: I tried the tutorial I linked above, and can confirm that it worked enough that I got an error on screen confirming that Chromium is indeed not supported on Raspberry Pi model B+

Using that tutorial as a basis, i now have a working version using Midori instead, which should work on all Raspberry Pi models.
It's currently working fine in Bullseye, But I'll test it next in Bookworm.
When I'm done i'll post a tutorial someplace.

lurch, 2024-10-22:
The Raspberry Pi Model B+ only has 512MB of RAM, and in my experience the latest versions of Chromium and Firefox really struggle to run with less than 1GB of RAM. So hopefully Midori will still be "lightweight" enough for these older Raspberry Pis.
Also, I think Firefox is only built for ARMv7 and above CPUs, and so can't run on the ARMv6 CPU in the Raspberry Pi Model B+.

lurch, 2024-10-22:
> I made a tutorial, I would appreciate if if someone can test it with later Pi versions.

That's really a question for the [forums](https://forums.raspberrypi.com/), not for GitHub :slightly_smiling_face:

qbicdesign, 2024-10-22:
why? this thread is specifically about the tutorial in the documentation