raspberrypi/documentation #3914: Document interrupts in pciex4_reset=0 mode

dsseng, 2024-11-08:
There is a firmware option `pciex4_reset` that could allow us to not reset the RP1-facing PCIe root complex upon transitioning to the kernel. This keeps RP1 peripherals memory-mapped and their registers are available for access by low-level software such as kernel or bootloader without PCIe or RP1 driver.

This works well for using UART, GPIO and similar peripherals on the RP1, but nothing is known about where do interrupts go, and these seem pretty much necessary to work with DWC3 or MACB peripherals in the RP1. Could you please document how are these interrupts configured by the firmware to access peripherals (at least DWC3 cores) or if not, how does the firmware manage to access USB3 storage without DWC3 interrupts. Thank you!

lurch, 2024-11-11:
I guess this is a request for extra information in https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf ?

ping @timg236 in case he's able to provide a quick answer to any of the firmware-related questions.

timg236, 2024-11-11:
The 2712 bootloader doesn't use any interrupts except for the VPU timer. It's all polled, async state-machines because the VPU doesn't have anything else to do except wait for IO :)

dsseng, 2024-11-11:
Not really, that's about how firmware maps the interrupts in its RP1 bring-up code, so whoever uses no reset could use the HW.

dsseng, 2024-11-11:
> The 2712 bootloader doesn't use any interrupts except for the VPU timer. It's all polled, async state-machines because the VPU doesn't have anything else to do except wait for IO :)

Oh, thanks for the fast reply! Will know, yet it would be great if it's explicitly noted in the docs for `pciex4_reset` like `MMIO is at offset 0x1f00000000, interrupts for RP1 hardware are disabled`.

closed 2025-01-26 by github-actions[bot]