raspberrypi/documentation #3915: The instructions for connecting a camera are misleading for Pi5

sandyol55, 2024-11-10:
In the "Accessories/Camera/Connect the Camera" section the current instructions are :-

_The flex cable inserts into the connector labelled CAMERA on the Raspberry Pi, which is located between the Ethernet and HDMI ports. The cable must be inserted with the silver contacts facing the HDMI port._

The 2nd is incorrect for Pi5, and neither are appropriate for the Zero  range.

Propose some re-wording referencing the alignment of the cable stiffener relative to the connector latch rather than the surrounding components, and perhaps an image as below.

Happy to have a go at a PR.

[image]

nathan-contino, 2024-11-11:
Agreed that this could use some clarification. I've found myself looking for this exact doc in the last couple of weeks, so I know it's not covered right now.

Please do open a PR with some fixes, even if it's a little rough. Happy to clean it up and get it over the finish line.

robertpeteuil, 2024-12-17:
Came here to report this issue.

I just wasted two hours troubleshooting my camera connection because all the docs/guides/videos are wrong for the Raspberry Pi 5.

nathan-contino, 2024-12-17:
We hear you, @robertpeteuil . I've experienced similar problems in the past. I'll take some better pictures of this tomorrow so we can get this fixed. For too long I've allowed perfect to be the enemy of good here.

lurch, 2025-01-27:
@nathan-contino The linked PR has now been auto-closed?

nathan-contino, 2025-01-27:
I'll be adding our own pictures that are more in line with our style guide. But we do appreciate the contribution!