raspberrypi/documentation #3923: Touch Display 2 overlay

foonerd, 2024-11-20:
The panel-ilitek-ili9881c.c which Touch Display 2 is based on already contains orientation and relevant parts are in the driver already:
https://github.com/raspberrypi/linux/blob/57b528e557890f25e010b6bc7356b5a716c79db2/drivers/gpu/drm/panel/panel-ilitek-ili9881c.c#L66

As such, overlay can be easily extended with "rotation=<value>" like many other panels do by appending rotation to:
https://github.com/raspberrypi/linux/blob/57b528e557890f25e010b6bc7356b5a716c79db2/arch/arm/boot/dts/overlays/vc4-kms-dsi-ili9881-7inch-overlay.dts#L120

```
		...
		disable_touch = <&gt911>, "status=disabled";
		rotation = <&dsi_panel>, "rotation:0";
```

Working dbto is attached to the community thread:

https://community.volumio.com/t/guide-official-raspberry-touch-display-2/69791#p-213411-using-official-raspberry-touch-display-2-with-volumio-os-3

Above simple solution removes need for cmdline.txt manipulation altogether.

lurch, 2024-11-20:
ping @pelwell and @6by9

6by9, 2024-11-20:
Ack that https://www.raspberrypi.com/documentation/accessories/touch-display-2.html#rotate-screen-without-a-desktop stating to use
```
video=DSI-1:1280x720@60,rotate=<rotation-value>
```
is wrong for the Touch Display 2, and it should be 720x1280.

Also 
```
sizex | Sets X resolution (default 1280)
sizey | Sets Y resolution (default 720)
```
are has the default values swapped.

I'm happy enough to add the parameters to the overlay, but want to do more checking as to which environments get updated by that before the docs get updated to recommend it.

6by9, 2024-11-20:
Note that including `display_auto_detect=1` with manually loading `dtoverlay=vc4-kms-ili9881-7inch` means that any fixups applied by the display auto detect (eg `dsi0`) will not be relayed into the manual dtoverlay. You therefore have the potential to end up configuring 2 displays accidentally when you wanted to change the rotation only.

foonerd, 2024-11-20:
@6by9 - Opened https://github.com/raspberrypi/documentation/issues/3922 for the documentation regarding Touch Display 2 Rotate screen without a desktop

We have two pieces as such (and apologies for the redundant links):
cmdline parameters - https://github.com/raspberrypi/documentation/issues/3922
overlay - this issue.

display_auto_detect=1 followed by dtoverlay=vc4-kms-ili9881-7inch,rotation=270 produces unified landscape experience from boot, throughout plymouth all way up to x11 session (xrandr rotated) on Bookworm. 

Use cases with multiple displays connected will need to be evaluated further.

closed 2024-11-21 by nathan-contino