raspberrypi/documentation #3952: Add the Raspberry Codec Zero mono-speaker screw terminals polarity

alxlve, 2024-12-07:
The Raspberry Pi Codec Zero documentation has a pinouts section with an image that describes the pinouts channels and polarity. The mono-speaker's screw terminals are undocumented.

Please add an image and a description about the polarity expected from these screw terminals, i.e.: is the left terminal the positive or the negative one, is the right terminal the positive or the negative one? This information is currently nowhere to be found.

The speaker polarity matters for more accuracy in the sound produced from the signal.

alxlve, 2024-12-07:
Link for the pinouts section in the documentation: [https://www.raspberrypi.com/documentation/accessories/audio.html#pinouts-4](https://www.raspberrypi.com/documentation/accessories/audio.html#pinouts-4)

lurch, 2024-12-07:
I just did a quick internet-search, and everything I found said that polarity only matters if you have two speakers, but if you only have one speaker it doesn't matter? :speaker:

alxlve, 2024-12-23:
It does matter even with a mono speaker.

closed 2025-03-02 by github-actions[bot]