raspberrypi/documentation #3956: "Specify a video output device" section is incorrect or incomplete (Rpi4b, Bookworm)

stbuerger, 2024-12-08:
The mechanism for specifiying the output HDMI connector isn't working as descriped in this section.
Either dependencies or environment influences or shown syntax makes (c)vlc to not behave as expected.
On a system with
```
kmsprint | grep Connector
Connector 0 (32) HDMI-A-1 (connected)
Connector 1 (42) HDMI-A-2 (connected)
Connector 2 (48) DSI-1 (connected) 
```

the option `--drm-vout-display [HDM-A-1|HDMI-A-2|DSI-1]` doesn't change the output device/port

As of 2024-12-08 the web is full of postings that report similar issues with (c)vlc and also for mpv and others.  So this seems to be a fundemenal problem of understanding or implementation of Rpi multi-monitor configurations.

Sofar the only way I found to display video on a specific output is ..
- run vlc with GUI
- goto menu "Tools" -> "Preferences" -> "Video" and define "Fullscreen Video Device" and SAVE
- quit vlc 
- start a new vlc instance with options `--fullscreen --play-and-exit `
In fact this only works with vlc and not with cvlc and same options

It really is a pitty that it is so complicated and irritating to make mult-monitor configurations work for kiosk or home-cinema use cases.

lurch, 2025-02-13:
@spl237 Is this another one of those "That's just how Wayland works" limitations?

@stbuerger I dunno if it'll help for your use-case, but I remember that when I was testing this a while ago the `cvlc --drm-vout-display` option **did** work correctly if I booted Raspberry Pi OS to the console instead of the graphical desktop.

spl237, 2025-02-13:
> [@spl237](https://github.com/spl237) Is this another one of those "That's just how Wayland works" limitations?

It sounds to me like a peculiarity of how VLC works with Wayland, and hence a VLC problem rather than one with our OS.

lurch, 2025-02-13:
@nathan-contino Perhaps https://www.raspberrypi.com/documentation/computers/os.html#specify-a-video-output-device just needs to clarify that it only works when booting to the console, and doesn't work from within the GUI desktop environment?

stevedavies-rp, 2026-09-10:
Thank you for this contribution @stbuerger. The documentation is now being updated and will be live soon.

closed 2026-09-10 by stevedavies-rp