raspberrypi/documentation #3985: Documentation "Use your Raspberry Pi as a network bridge"

spicavirgo9, 2024-12-14:
I experienced a problem when following the instructions in
[https://www.raspberrypi.com/documentati ... ork-bridge](https://www.raspberrypi.com/documentation/computers/configuration.html#use-your-raspberry-pi-as-a-network-bridge)
using a RasPi4B with Bookworm in headless mode, connected with a ethernet cable to my internet-router.

After the command "sudo nmcli connection up Bridge" the bridge did not work. The cause was, that the device eth0 was still occupied by the usual "Wired connection 1" connection.
To make the bridge work, the command "sudo nmcli connection down 'Wired connection 1' " has to follow. If the connection to the Raspi used the Ethernet port (e.g. in headless mode or via ssh), this command will disrupt it and it has to be reestablished.

I would appreciate a update of the instructions on the above mentioned website.

lurch, 2024-12-14:
I've never tried doing this myself, but if you're setting up a wireless bridge on your Pi (which will presumably involve configuring both the Ethernet and Wifi interfaces), it seems a bit "risky" to try doing that on a headless Pi, rather than with a monitor and keyboard temporarily attached? (because as you say, it's easy for any SSH connection to get disrupted) :thinking:

slain2, 2024-12-14:
Tested it using a headless Pi4 connected via Ethernet, Bookworm 64-bit. The Hotspot does not work until after a reboot (i.e. a mobile phone cannot connect to it.)
Before reboot: nmcli dev
DEVICE         TYPE      STATE                                  CONNECTION
eth0           ethernet  connected                              Wired connection 1
lo             loopback  connected (externally)                 lo
wlan0          wifi      connected                              Hotspot
bridge0        bridge    connecting (getting IP configuration)  Bridge
After reboot (phone connects OK):
nmcli dev
DEVICE         TYPE      STATE                   CONNECTION
bridge0        bridge    connected               Bridge
lo             loopback  connected (externally)  lo
eth0           ethernet  connected               Ethernet
wlan0          wifi      connected               Hotspot

nathan-contino, 2024-12-15:
Thanks so much for this feedback. I'll get this fixed on the docs page ASAP. nncli can be a tricky beast, I wonder what's changed since I originally wrote and tested this section?

Anyway, glad to hear you found a working solution. Even if the docs had an issue, at least we got you close enough to figure out the rest. Hopefully with this change the next user will have an even easier time.

spicavirgo9, 2024-12-15:
Thanks for your comments. Glad to contribute a quantum to the RasPi community.

spicavirgo9, 2024-12-16:
@nathan-contino Another hurdle I found in the instructions is: The password that was set for the hotspot gets erased by the command "sudo nmcli connection modify 'Hotspot' master bridge0".
That leads to an error message when activating the hotspot because of the now missing password.

mickjc750, 2025-01-04:
I was unable to get this working.

- Raspberry Pi 3 Model B Rev 1.2
- Flashed a virgin image of 'Pi OS Lite (64 bit)' with Raspberry Pi imager.
- Set my country to Australia in raspi-config to allow wifi to work.
- Exactly followed these instructions starting from 'Use your Raspberry Pi as a network bridge' [https://www.raspberrypi.com/documentation/computers/configuration.html#use-your-raspberry-pi-as-a-network-bridge](url)

As instructed, I did not execute 'sudo nmcli connection modify' as I had not created a hotspot in the above instructions, I instead executed the 'sudo nmcli connection add con-name 'Hotspot'', replacing \<hotspot-ssid\> with my ssid and \<hotspot-password\> with my password (both lowercase alpha chars).

On completion nmcli device showed:
```
DEVICE         TYPE      STATE                                  CONNECTION         
eth0           ethernet  connected                              Wired connection 1 
lo             loopback  connected (externally)                 lo                 
wlan0          wifi      connected                              Hotspot            
bridge0        bridge    connecting (getting IP configuration)  Bridge             
p2p-dev-wlan0  wifi-p2p  disconnected                           --             
```
And after a reboot 
```
DEVICE         TYPE      STATE                   CONNECTION 
bridge0        bridge    connected               Bridge     
lo             loopback  connected (externally)  lo         
eth0           ethernet  connected               Ethernet   
wlan0          wifi      connected               Hotspot    
p2p-dev-wlan0  wifi-p2p  disconnected            --        
```

The SSID appears on my laptop and phone, but the password is not accepted on either.
The password is just 8 lower case alpha characters.

spicavirgo9, 2025-01-04:
Your output of `nmcli device` looks good for me. Can you make a access point without bridge working on your RasPi?

Please try `nmcli device wifi show-password`.
Does this show the expected password? If not, set the password for the hotspot again by using the GUI (Grafic User Interface).

mickjc750, 2025-01-04:
Hi, thanks for responding, so the first thing i tried was `nmcli device wifi show-password` and yes, it showed the correct ssid passowrd and a QR code. 
 I then re-flashed a new image to start from scratch, chose my country to enable wifi, then executed the `sudo nmcli device wifi hotspot ssid <example-network-name> password <example-password>`
 The normal hotspot has the same problem, I can see the SSID but can't connect.

 This is a very old Pi, which makes me think it might be a firmware problem. 
Checking my firmware version:
```
Aug 30 2024 19:19:11 
Copyright (c) 2012 Broadcom
version 2808975b80149bbfe86844655fe45c7de66fc078 (clean) (release) (start)
```
executing `sudo rpi-update` upgraded this to `e4c86cd7b1a1291368ac0d2aaa3534a1d6beb3d5` But this hasn't solved the problem.

It does work as a regular hotspot if I run debian 11 and follow some instructions given by chatgpt using hostapd 
 and dnsmasq. But chatgpt failed to make it work as a bridge. 

 The reason I want it to bridge, is that I have a network camera (consisting of a PiZeroW + cam), slightly out of my routers wifi range. So I wanted to connect it via this Pi3, but I need it visible on my lan to access it as a network camera. As the camera is a Pi, it might be possible for it to ssh into m motioneye server and forward the port, but with ssh encrypting the data it may not have enough performance.

spicavirgo9, 2025-01-05:
I have a fabric new RasPi4B but was unable to connect a Espressif ESP32-S3 based device with the hotspot of the RasPi. Then I used a very old USB WLAN stick with this RasPi and it worked - hurray!
Wish you good luck with your project.

closed 2025-03-16 by github-actions[bot]

domisol, 2025-05-04:
My bridged connection works. What I did is modifying the current Ethernet connection with:
`sudo nmcli connection modify "Ethernet" slave-type bridge con-name 'Ethernet' ifname eth0 master bridge0`
I rebooted also after `the sudo nmcli connection add con-name 'Hotspot'     ifname wlan0 .........`

`nmcli dev
DEVICE         TYPE      STATE                   CONNECTION 
bridge0        bridge    connected               Bridge     
lo             loopback  connected (externally)  lo         
eth0           ethernet  connected               Ethernet   
wlan0          wifi      connected               Hotspot    
p2p-dev-wlan0  wifi-p2p  disconnected            --         
`