raspberrypi/documentation #3986: New raspberry pi documentation look?

steveiliop56, 2024-12-14:
I just opened the raspberry pi documentation to check something and I was greeted by this:

[image]

Is this some sort of update? To be honest it looks really bad compared to the previous UI, it looks like it lost all of its styles. The dark mode is cool but the SVG for the theme change looks really bad, the `color-scheme` is not set resulting in white scrollbars and the search remains white. Hopefully this is some sort of accidental config change because the old one looked so much better.

nathan-contino, 2024-12-15:
We've made some tweaks to the site layout to make it easier to navigate between all of the different site sections. The previous layout chopped up the site into smaller pieces (computers, microcontrollers, accessories, etc) with entirely separate nav. This frequently confused users, who would wind up in one section but misunderstand how to find deeply nested content in a separate section (for instance, finding 'Raspberry Pi OS' from Accessories). This makes those jumps and discoverability easier, but longer term we'd like to improve that organisation even more, and the previous tabbed system only scaled to a small number of tabs. That's to say nothing of the massive strides we've made in mobile navigability.

I'm sorry to hear that you don't like the aesthetics, but "looks really bad" is a subjective statement, not an objective problem worthy of a GitHub issue.

If you have concrete, specific problems with the new site appearance that you'd like to see changed, we're happy to accept PRs and feedback. I'm well aware that nothing is perfect and we can always improve things. But this issue isn't helpful.

FWIW we've seem overwhelmingly positive feedback from users so far, so I suspect (and I suppose personally hope?) you simply differ from the majority opinion here. I know it can be jarring when a site you frequently use changes styling, but hopefully the new style grows on you. It will make it much easier to grow and improve the documentation in the future, and the content is the most important part, I hope you can agree.

closed 2024-12-15 by nathan-contino

steveiliop56, 2024-12-15:
Sorry I was a bit harsh. Let me be more specific. I get the layout change and since it has a solid reason behind it I agree. What I don't like is that the navbar looks like it "lost" if all it's styles, by this I mean that it now looks like a simple html component, I really like the more modern approach you are trying to achieve but I don't think the current jekyll theme can handle this change. The ideal approach for me would be a theme/navbar like [nextra](https://nextra.site/) which basically is the docs theme you are trying to achieve. Another issue is that it suddenly lost all animation, to be more specific, here is a screenshot of the mobile UI:

[image]

You can see that the hamburger menu looks much better than the docs one and when I click the upper one I am greeted with a cool animation of the button changing style and the menu appearing that looks really nice. On the other hand the hamburger menu and theme change of the docs are using not that good SVGs, you can use icons from [iconify](https://iconify.design) that are much more polished than the ones you are using now. Additionally the docs lack any animation, whatever I click happens instantly and especially the dark mode switch is really bad when it doesn't have an animation. Concerning the navigation menu, I think that a side drawer (like the one the inconify website uses) would look much better than a full screen pop up. 
Also about the dark theme, when you switch the theme you should set the color scheme in the root of the html so as browsers like chrome switch the scrollbar color from light to dark.
Last but not least the search bar is not that great, firstly it doesn't switch theme while it seems like the text inside it goes white so it's basically an empty search bar in dark mode (see screenshot) and again a full screen pop up just to search seems a bit to much but I don't have an issue with it if you prefer it.

A thing that I also don't like that much but it's not an issue is that the dark theme is just #000000 which is a bit too black. By this I mean that it would be better to use a slightly lighter shade of black, something more gray-ish instead of pure black.

While all these can be achieved with the current jekyll setup I don't think it can handle your modernization idea going forward. I would recommend trying other alternatives that are designed to make modern looking websites like [nextra](https://nextra.site), [docusaurus](https://docusaurus.io/), [vitepress](https://vitepress.dev/) or even [astro](https://astro.build/).

Hopefully I was much more helpful now.

steveiliop56, 2024-12-15:
To add to the previous comment, if you are interested in switching framework to something else I would be happy to try out the alternatives I mentioned above with your current docs site and see what's more suitable and open a PR.

nathan-contino, 2024-12-15:
Thanks so much for the detailed response. I agree with most of this! Our docs resources a bit low at the moment, but I'll look into some of those options (iconify would be a very easy improvement, transitions are fairly quick to add).

Ultimately I would like to move to one of those solutions you listed (I especially like docusaurus and have some prior positive experience with it) or simply use more of asciidoctor's built-in styles. But right now our docs hierarchy is so deeply nested it would test the limits of Asciidoctor, so this was a smaller immediate improvement in the spirit of iterative change.

I would welcome any improvement PRs you have the time to make. Might have to do some infrastructure work to preserve and redirect existing external and internal links, but I think it's very much worth it.

In the short term, I'll create some issues for these improvements to better track them. Thanks again!

steveiliop56, 2024-12-15:
Cool! I will experiment with these solutions to see what would look the best with rpi docs.