raspberrypi/documentation #4002: Make Pi5's MIPI connector CSI/DSI distinction abundantly clear on relevant page(s)

earcam, 2025-01-10:
Was going to comment on https://github.com/raspberrypi/documentation/issues/3155 but, as this a documentation project, think the distinction warrants a seperate issue.

Vendors such as pimoroni and pihut are selling the DSI cables, and make it clear on their various product pages (inc CSI).

But idiots like me, who've a bunch of Pi-Zero camera MIPI cables, _may_ waste time when migrating a setup from Pi 4 to 5;  checking the product page and searching "DSI" in [docs sidebar](https://www.raspberrypi.com/documentation/) didn't yield anything obvious.

Please add a banner note to the [TouchScreen docs page](https://www.raspberrypi.com/documentation/accessories/display.html), something like "Note: for Raspberry Pi 5, the CSI and DSI cables are different, mini-CSI cables will work for cameras, but Pi5 to touchscreen requires a specific cable".

... and If I've bricked my touchscreen, please prioritise this issue!

nathan-contino, 2025-01-10:
Thanks for noting this. Have you taken a look at the [Touch Display 2](https://www.raspberrypi.com/documentation/accessories/touch-display-2.html) docs? Those make a clearer distinction between the 22- and 15-pin (mini and non-mini, respectively) cables you need for each Pi model.

Unfortunately we originally wrote the Touch Display page before the Pi 5 or Zero existed, so we didn't account for that disambiguation. I think we should adopt some of the language and instructions from the TD2 page into the TD page.

lurch, 2025-05-18:
I'm going to close this, and fold it into #4097

closed 2025-05-18 by lurch