raspberrypi/documentation #4003: Update needed to "Raspberry Pi Debug Probe" for PICO 2

cpottle9, 2025-01-11:
I see two issues that need to be addressed.

1) There is no PICO2 H. Need to describe manually installing headers.

2) The openocd version available on the PI 4 bookworm does not support the 2350.
    I eventually found the repo https://github.com/raspberrypi/openocd/tree/sdk-2.0.0 and built openocd.
    Not trivial. 
    You need to either provide a detailed recipe or provide downloadable binaries.

cpottle9, 2025-01-13:
Someone pointed out on Raspberry PI Forums that instructions for building openocd for PICO 2 is are included in Appendix A of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf.

I think that information belongs with the rest of the Debug Probe documentation here: https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html.

XECDesign, 2025-01-14:
I haven't heard anything from any maintainer of our OpenOCD fork asking for the package to be updated, so I assume it's not ready yet.

cpottle9, 2025-01-14:
I guess the process for updating openocd for raspberry pi os would include upstreaming the RP2350 changes to the main openocd repo and then waiting for an official version.
That is the only way to ensure all the other hardware targest are supported.

In the meantime I think it would be sufficient to update the debug probe documentation providing a link to appendix A of "Getting started with Raspberry Pi".

I wasted a couple of  finding raspberry pi openocd repo and figuring out how to build it.
This small change would have helped me immensely.

Lastly, I agree with issue https://github.com/raspberrypi/documentation/issues/3831.
A photo of a PICO 2 and PICO W with the 3 pin header soldered in place and the cable connected.

lurch, 2025-05-11:
https://github.com/raspberrypi/debugprobe/releases/ contains [`debugprobe_on_pico2.uf2`](https://github.com/raspberrypi/debugprobe/releases/download/debugprobe-v2.2.2/debugprobe_on_pico2.uf2) and https://github.com/raspberrypi/pico-sdk-tools/releases contains [`openocd-0.12.0+dev-aarch64-lin.tar.gz`](https://github.com/raspberrypi/pico-sdk-tools/releases/download/v2.1.0-0/openocd-0.12.0+dev-aarch64-lin.tar.gz) - does that meet your requirements?

@P33M and @will-v-pi - is there any reason why the version of openocd in our APT repos can't / shouldn't be updated?

cpottle9, 2025-05-11:
@lurch, my issue is not about the debugprobe*.uf2 files.
I identified two issues in my original text.

The first is I did not see any documentation on using the debug probe with a  Pico other than a Pico H.
I now see a note in [Raspberry Pi Debug Probe#Getting Started](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/microcontrollers/debug-probe/getting-started.adoc) that describes soldering header pins.
That note has been there for quite a while. I missed that when reading the document. Bad me....
Having said that, adding a picture of a pico with the header installed would be helpful.

The second issue is how to use acquire an openocd version for rp2350 and how to use it.
That is documented in one place only, appendix A of [Getting Started with Pico and ...](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).

Rightly or wrongly, I assumed everything I need to know to use the debug probe would be included in [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html).
At a minimum, there should be a reference in **Raspberry Pi Debug Probe** to appendix A of the **Getting Started...** document.

I agree the raspberry pi apt repos should include a version of openocd that support the rp2350.

lurch, 2025-05-12:
> At a minimum, there should be a reference in **Raspberry Pi Debug Probe** to appendix A of the **Getting Started...** document.

There is 🙂  (although it's admittedly a bit buried away and should probably be more prominent).

[image]

Given that https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#install-tools has now drifted **so** far out of sync with Appendix A of the "Getting started with Raspberry Pi Pico-series" PDF, I wonder if it might be better to just delete those instructions entirely and _only_ refer to the GettingStarted Appendix? That way we'd only need to keep the instructions up to date in one place. 🤔  WDYT @JamesH65 and @mudge ?

cpottle9, 2025-05-12:
Before I get into my detailed response to your most recent comment:
- Using GDB to debug is not a 'getting started' thing. From my coding experience (over 40 years) I can tell you many developers get through their entire careers debugging with print statements.
- Clearly, the portal [Raspberry Pi Documentation](https://www.raspberrypi.com/documentation/) is intended to be a user friendly entry point the documentation. The section for debug probe could be much better and moving content from it to the appendix of pdf will make it less user friendly.
- In section '4.3 Debug' of [Getting started with Raspberry Pi Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) I see a link to [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html
). Circular references in documentation is not ideal. You folks need to decide which document is definitive.

@lurch, the text you captured is clearly in the section **Install GDB**. 
And the placement of the sentence makes it look like it applies to MS Windows only.
Further, the text is incorrect in that **Getting Started with Raspberry Pi Pico** does not have a chapter 9.

Scanning through [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) I also see a reference to chapter 6 of **Getting Started with Raspberry Pi Pico** .
There are only 5 chapters.

I would suggest [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html)  needs significant rework to include RP2350 and other details.