raspberrypi/documentation #4004: Raspberry Pi 5 recommended power supply inconsistent

fred-c1, 2025-01-12:
https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/power-supplies.adoc
Raspberry Pi 5   => 5A        (with 1.6A for USB)    =>    5-1.6=3.4A for the board
Raspberry Pi 5   => 3A        (with 0.6A for USB)    =>    3-0.6=2.4A for the board

2.4A or 3.4A for the board ????

nathan-contino, 2025-01-14:
@timg236 any suggestions here? I have to agree that the math doesn't check out, but maybe I'm missing something. Would be nice to add an explanation for the curious.

timg236, 2025-01-14:
It depends on whether the USB current limiter is set to the high or the low limit

timg236, 2025-01-14:
That's correct. So, if someone was to blindly override the USB current limiter on Pi5 with a 3A power supply then USB can consume up to 1.6A (e.g. a hard driver was attached) leaving only 1.4A available for everything else which might crash if a CPU stress test was run.
The purpose of the current limiter is to avoid power hungry USB devices from using so much power that the board crashes under load.

fred-c1, 2025-01-15:
About "It depends on whether the USB current limiter is set to the high or the low limit":

Does this mean that the board will reduce it's adjust / performance (CPU clock frequency I guess ??)  (based on the current capability reported by the power supply on the CC pin of the USB C connector and also based on the USB current limiter setting) ?

timg236, 2025-01-15:
No. The CPU is frequency is not limited according to the max-current claimed by power supply. The only lever is whether to select the high or the low limit for the downstream devices.

lurch, 2025-01-15:
But I believe you can manually "underclock" the CPU (i.e. tell it to run at a lower frequency), if you want it to use (slightly?) less power https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options
And a quick search finds https://forums.raspberrypi.com/viewtopic.php?t=376416
Any additional questions you have here should probably also be asked on the forums rather than GitHub, so I'll close this issue.

closed 2025-01-15 by lurch

fred-c1, 2025-01-15:
Sorry but, if it is not the CPU clock frequency, then what makes the difference between "3.4A for the board" and "2.4A for the board"?

fred-c1, 2025-01-15:
So the power supply should be able to provide how much current for the Pi itself (no USB connected) (without over or under cloking)?

fred-c1, 2025-01-15:
Thanks: I go with this answer (3A)
I was just trying to clarify this on the doc. page.