raspberrypi/documentation #4007: Pi SSD installation

Gordon999, 2025-01-18:
https://www.raspberrypi.com/documentation/accessories/m2-hat-plus.html#m2-hat-plus-installation

OK as far as fitting it, but then if it's not partitioned you can't get it to mount at boot etc.

Needs more information on partitioning, mounting and formatting for those not familiar with doing it.

lurch, 2025-01-18:
> OK as far as fitting it,

@nathan-contino I followed these instructions myself a couple of weeks ago, and IMHO the bit where it says "Disconnect the ribbon cable from the M.2 HAT+" should probably be elaborated to mention lifting up the cable-holder before trying to remove the FFC cable?

>  if it's not partitioned you can't get it to mount at boot etc.

Further down that page it _does_ mention writing an OS to the SSD using Raspberry Pi Imager https://www.raspberrypi.com/documentation/accessories/m2-hat-plus.html#boot-from-nvme
Although @nathan-contino perhaps it should say "Flash an OS your NVMe drive using Raspberry Pi Imager." rather than "Format your NVMe drive using Raspberry Pi Imager."? Also, it should probably mention the [Network Installer](https://www.raspberrypi.com/documentation/computers/getting-started.html#install-over-the-network) option? (as that allows you to write an OS to the SSD _without_ using an SD card)

> Needs more information on partitioning, mounting and formatting for those not familiar with doing it.

I believe that these are just "generic Linux instructions", which there are many third-party guides for; whereas the documentation at https://www.raspberrypi.com/documentation/ is _mainly_ focused on Raspberry-Pi-specific things which aren't documented elsewhere?

Obviously if you're writing an OS to the SSD using Raspberry Pi Imager, then the separate "partitioning, mounting and formatting" steps aren't necessary. I wonder how much of a niche use-case it is to have an SSD on your Pi but **not** be booting an OS from it? 🤔

Gordon999, 2025-01-18:
Section 10 Says 

Congratulations, you have successfully installed the M.2 HAT+. Connect your Raspberry Pi to power; Raspberry Pi OS will automatically detect the M.2 HAT+. If you use Raspberry Pi Desktop, you should see an icon representing the drive on your desktop. If you don’t use a desktop, you can find the drive at /dev/nvme0n1. To make your drive automatically available for file access, consider [configuring automatic mounting](https://www.raspberrypi.com/documentation/computers/configuration.html#automatically-mount-a-storage-device).

Well at this stage there was no icon and you can't configure to automatically boot as it's not partitioned or formatted, so sudo blkid fails to find it.

I'm new at this and it confused me...

lurch, 2025-01-18:
Ahhh, I guess those instructions were written with the assumption that the SSD you're using is already partitioned and formatted (which AFAIK **is** always the case for microSD cards and USB flash drives). So yeah, the documentation probably needs some clarification in the event that the SSD is "totally blank".

Gordon999, 2025-01-18:
Yes l bought the raspberrypi ssd kit, the ssd is blank.

stevedavies-rp, 2026-09-16:
Thanks for raising this issue, @Gordon999. We are editing the documentation to address this and it will be live soon.

closed 2026-09-16 by stevedavies-rp