raspberrypi/documentation #4033: "display_auto_detect" is not enabled by default on Pi 5

spockfish, 2025-02-19:
### Describe the bug

The documentation states that `display_auto_detect` is on by default. However, on the Pi 5 it is not.

### Steps to reproduce the behaviour

Without *any* graphics overlay configured, with `display_auto_detect` on, the touchscreen driver is loaded. So for an official Raspberry Pi touchscreen this is the `raspberrypi_ts` module. For the the official Display 2 this is the `goodix_ts` module.

However, on a Pi 5 this is not happening, unless `display_auto_detect=1` is explicitly set in `config.txt`, suggesting that it is `0` by default. 

### Device (s)

Raspberry Pi 5

### System

6.6.78

lurch, 2025-02-19:
> the documention does not mention a specific distribution anywhere.

The entirety of https://www.raspberrypi.com/documentation/ is written to target the latest version of Raspberry Pi OS. If you're using a 3rd-party OS, then you'll have to check the documentation for that other OS.

@nathan-contino Similar questions / issues have cropped up before, so perhaps this should be explicitly stated in https://github.com/raspberrypi/documentation/blob/develop/README.md ?

popcornmix, 2025-02-19:
I agree the current wording is imprecise:
```
Enabled by default.

When enabled, the firmware will automatically load overlays for recognised DSI displays.

To disable, set display_auto_detect=0.
```

As a config.txt option (where this is being documented) it is not enabled by default.
To disable it you just remove the line (`display_auto_detect=0` is a no-op).

closed 2025-05-08 by lurch