raspberrypi/documentation #4046: Incorrect Documentation for requiring sudo Password

masselmello, 2025-03-12:
In the [documentation](https://www.raspberrypi.com/documentation/computers/configuration.html#require-a-password-for-sudo-commands) for requiring a password for sudo commands it mentions that I have to edit the `010_<username>-nopasswd` file by adding this line:

`<username> ALL=(ALL) PASSWD: ALL`

The issue is that my installation does not have this file. Instead I have a file called `010_pi-nopasswd`. This already has the following entries:

`<username> ALL=(ALL) NOPASSWD: ALL`

To my understanding this should not be the case according to the documentation. The file should not have the pi user name in it but the username of the user that I created when flashing the image with Raspberry Pi Imager. 
The username I created is definitely very different to "pi".

If I change the values in the `010_pi-nopasswd` file to `<username> ALL=(ALL) PASSWD: ALL` it also works as intended and requires me to type my password for sudo. But if I delete the `010_pi-nopasswd` file and instead add the correct file `010_<username>-nopasswd` with `<username> ALL=(ALL) PASSWD: ALL` as it's content. 
Then I still do **not** need to type my password for sudo. **It has no effect.**

Is this a mistake in the documentation and the file should still be called `010_pi-nopasswd` and not `010_<username>-nopasswd` or is it a bug in the current Raspberry Pi OS Lite?

Here a screenshot of the documentation for quick reference:

[image]

lurch, 2025-03-12:
As it says in `/etc/sudoers.d/README`: "This will cause sudo to read and parse any files in the /etc/sudoers.d directory that do not end in '~' or contain a '.' character."
So it doesn't actually matter what the file is called. _However_ you might need to do a reboot for `sudo` to pick up any newly-added or renamed files? Does that fix the "It has no effect." that you were seeing?

> Instead I have a file called `010_pi-nopasswd`.

Yeah, it probably makes sense to rename that to match the username, just for consistency. Is that an easy fix @XECDesign ?

masselmello, 2025-03-12:
Ah I see. But I actually tried a reboot and also renamed the file to match my username. But as soon as I do that it has no effect anymore even after a reboot. Sudo won't ask for my password anymore. When I renamed it to match the pi-user it did have an effect.

So I assumed that the mistake is in the name of the file.

I will try to reflash the SD card when I am home this evening and try it again. Maybe something on my installation is just broken.

lurch, 2025-03-12:
Seems to be working as expected for me? I've set things up as you've described, and I get asked for a sudo password after a reboot:
```console
pipi@raspberrypi:~ $ ls -l /etc/sudoers.d/
total 24
-r--r----- 1 root root   36 Apr 29  2019 010_at-export
-r--r----- 1 root root   44 Jun 19  2024 010_dpkg-threads
-r--r----- 1 root root   31 Jul 25  2023 010_global-tty
-r--r----- 1 root root   27 Mar 12 10:52 010_pipi-nopasswd
-r--r----- 1 root root  211 Mar 18  2022 010_proxy
-r--r----- 1 root root 1096 Jun 27  2023 README
pipi@raspberrypi:~ $ sudo cat /etc/sudoers.d/010_pipi-nopasswd 
[sudo] password for pipi: 
pipi ALL=(ALL) PASSWD: ALL
```

masselmello, 2025-03-14:
So I also tried flashing the image again. The file is still named `010_pi-nopasswd` but creating a new one called `010_<username>-nopasswd` now works. Maybe there was something wrong with the installation.

But then the documentation is still wrong IMHO because the file that the system created is not called `010_<username>-nopasswd` but instead `010_pi-nopasswd`. So the documentation should be revised here to reflect the actual file structure. Otherwise it could lead to confusion on why there is still a file with the pi username but the documentation says it should have the username that was initially created by the user.

XECDesign, 2025-03-14:
> Yeah, it probably makes sense to rename that to match the username, just for consistency. Is that an easy fix [@XECDesign](https://github.com/XECDesign) ?

Fix for what and consistency with what? The file belongs to a package, so renaming it would be a bad idea. Since the filename has any relation to the username, I don't see the value in tinkering with it.

masselmello, 2025-03-14:
The file currently created by the system has no relation to the actual username. It is always named  the same way on my fresh install of Raspberry Pi OS.

`010_pi-nopasswd` 

Not like the documentation mentions it.
The documentation mentions that the file should contain the username. It does not do that. That was the actual confusion I wanted to report in this issue in the first place.

lurch, 2025-03-17:
Ahhh, sorry @XECDesign , I didn't realise that the file belonged to a package, rather than "just" being automatically created at first boot 🤦 

@masselmello As we're all in agreement, would you like to submit a PR with the appropriate change to the documentation?

masselmello, 2025-03-17:
@lurch Sure

closed 2025-04-23 by lurch