raspberrypi/documentation #4047: building offline documentation not completely offline

pfelecan, 2025-03-13:
Hello,

I built the documentation ant everything works well.
For the moment, there is only an issue with some documentation links which directs to the online site; e.g. in the `Schematics and mechanical drawings` section, all those referred documents should be in the local version, isn't it ?
I understand that there are links not in the raspberry domain/scope/ownership but the above mentioned documents could be in this repository.

lurch, 2025-03-13:
I assume you're talking specifically about https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#schematics-and-mechanical-drawings ?
That's intentional - those files are hosted at https://datasheets.raspberrypi.com/ as they're managed via a separate CMS system, and we won't be adding them to GitHub / git. You're free to download them locally if you wish, as long as you adhere to https://datasheets.raspberrypi.com/licence.html and https://www.raspberrypi.com/licensing/

closed 2025-03-13 by lurch

pfelecan, 2025-03-13:
OK. Thank you.