raspberrypi/documentation #4048: Code duplication in the chapter “Wiring peripherals”

Voss-Leonard, 2025-03-14:
Possible error [in this chapter](https://www.raspberrypi.com/documentation/computers/compute-module.html#examples).

This chapter really helped me understand how to set up devices on my Compute Module using just a few commands. 
Thanks, I'd love to see more!

It seems as if a command from the first example has slipped into the second. In any case, the command to display the current hardware time appears in both examples, even though the second example involves a completely different type of sensor—namely, an SPI Ethernet controller!

[image]

The lines outlined in red therefore make little sense to me. Or does this refer to the previous example? Please clarify and correct if necessary.

lurch, 2025-03-17:
ping @pelwell and @dp111

pelwell, 2025-03-17:
Surely you can manage this one on your own, @lurch?

lurch, 2025-03-17:
I don't have a Compute Module nor a ENC28J60 module with which to test things, nor am I overly familiar with DeviceTrees and DT-overlays...

pelwell, 2025-03-17:
This is about spotting an obvious cut-and-paste error in the documentation - Ethernet interfaces are not real time clocks.

commit 13a7f6ac (2025-04-23): Fix #4048
commit 9de619f8 (2025-04-23): Merge pull request #4053 from raspberrypi/lurch-patch-1

Fix #4048
closed 2025-04-23 by lurch