raspberrypi/documentation #4051: RTIMULibDrive does not build as indicated in documentation

swpalmer-cl, 2025-03-16:
Instructions here: https://www.raspberrypi.com/documentation/accessories/sense-hat.html#install
indicate that :

-  RTIMULibDrive, RTIMULibDrive10, and RTIMULibDrive11  should build with `make`
- RTIMULibDrive11 is pre-compiled

Both of these are wrong.
Following the instuctions on that page you will find thatRTIMULibDrive11 is not pre-ciompiled and that any attempt to `make` any those three projects results in the following linker errors:
```
RTIMULibDrive.cpp:(.text+0x34): undefined reference to `RTIMUSettings::RTIMUSettings(char const*)'
/usr/bin/ld: RTIMULibDrive.cpp:(.text+0x40): undefined reference to `RTIMU::createIMU(RTIMUSettings*)'
/usr/bin/ld: RTIMULibDrive.cpp:(.text+0xe8): undefined reference to `RTMath::currentUSecsSinceEpoch()'
/usr/bin/ld: RTIMULibDrive.cpp:(.text+0x18c): undefined reference to `RTMath::currentUSecsSinceEpoch()'
/usr/bin/ld: RTIMULibDrive.cpp:(.text+0x1cc): undefined reference to `RTMath::displayDegrees(char const*, RTVector3&)'
collect2: error: ld returned 1 exit status
make: *** [<builtin>: RTIMULibDrive] Error 1
```

lurch, 2025-03-17:
> * RTIMULibDrive, RTIMULibDrive10, and RTIMULibDrive11  should build with `make`

I just tested this on Raspberry Pi OS 64-bit on a Raspberry Pi 5. I cloned the https://github.com/RPi-Distro/RTIMULib repository, `cd`-ed to each of `RTIMULib/Linux/RTIMULibDrive`, `RTIMULib/Linux/RTIMULibDrive10` and `RTIMULib/Linux/RTIMULibDrive11` in turn, and in each directory I was successfully able to run `make`. I got a bunch of compiler warnings, but the build completed without errors.

Can you think of what you might have done differently that means that it doesn't work for you? Do you have the `librtimulib-dev` APT package (which is one of the dependencies of the `sense-hat` APT package) installed?

> * RTIMULibDrive11 is pre-compiled

I did a bit of poking around, and the `sense-hat` APT package recommends the `librtimulib-utils` package, and the latter includes `/usr/bin/RTIMULibDrive11` .

swpalmer-cl, 2025-03-17:
Sorry I forgot to include my environment details:

- Raspberry Pi 3 Model B Rev 1.2
- Raspbian GNU/Linux 11 (bullseye)

Everything updated via `apt` including `librtimulib-dev`  and `librtiumlib-utils` (7.2.1-6).  Sorry, I do have RTIMULibDrive11 in `/usr/bin`, but when the page I referenced said it was pre-compiled I was thinking it would be found in the folder it mentioned.

You are not following the instructions on the page.  You cloned the GitHub repository.  While the GitHub library is hyperlinked, it is never actually mentioned. The instructions only say that the example code can be found at `/usr/src/sense-hat/examples` and "The C/C++ examples can be compiled by running make in the appropriate directory."

If the "sample code" isn't included in a buildable form perhaps it is better to not include it at all and refer to the GitHub repository instead.