raspberrypi/documentation #4060: Power Supply Docs incorrect

tomjdickson, 2025-03-28:
File: `documentation/asciidoc/computers/raspberry-pi/power-supplies.adoc`

On the Raspberry Pi website and on the Pi 5 box, the recommended voltage requirement for the Pi is 5V but the document referenced above is 5.1V.

JamesH65, 2025-03-28:
The Pi itself requries 5v AT THE PI. Our own power supply is 5.1V to compensate for any losses in the cable, so by the time the power gets to the Pi, the voltage has dropped to 5v.

closed 2025-03-28 by lurch

tomjdickson, 2025-03-29:
Thanks for the info. 

Considering this device is designed for beginners & hobbyists. Would it make sense to make this more clear? I don't think the target demographic for these devices will realise this.

lurch, 2025-03-30:
I think we'd expect beginners to buy one of https://www.raspberrypi.com/products/power-supply/ ?

And I think non-beginners would understand that 5.1V is pretty close to 5V anyway; and USB power supplies are always 5V (+/- 5%) ? 🤷‍♂

tomjdickson, 2025-03-30:
> I think we'd expect beginners to buy one of https://www.raspberrypi.com/products/power-supply/ ?

No, there is data on this from retailers. For example if you buy an in Australia, the main brick and mortar retailer doesn't even carry this. They recommend an incorrectly sized power supply.

lurch, 2025-03-31:
> For example if you buy an in Australia, the main brick and mortar retailer doesn't even carry this. They recommend an incorrectly sized power supply.

Interesting... could you email me some details please (to `<my firstname>.<my lastname>@raspberrypi.com`) and I'll ask our commercial team to investigate. Thanks.