raspberrypi/documentation #4063: Excessive number of DOM elements in table of contents

mudge, 2025-03-31:
@JamesH65 flagged that the [Hardware page](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html) is extremely slow in his browser even going so far as to crash it.

Looking at the structure of the page, it has over 50,000 DOM elements (for reference: Google Lighthouse will error if you have more than 1,400 elements on a page). While we'd expect a lot of elements as the Hardware section of the SDK is very long, looking at the table of contents alone reveals 12,858 elements and this appears on every single page. Digging a bit more, the table of contents seems to have 1,242 entries when fully expanded so that's the absolute minimum number of elements we would need to render it.

We can look at simplifying the markup for the table of contents but can we actually reduce the depth the table of contents goes to, e.g. to top-level sections only (e.g. "Getting started with your Raspberry Pi") rather than sub-sections (e.g. "Getting started with your Raspberry Pi - Power supply")? The theory being that would drastically cut down the size of the table of contents without also rewriting its markup.

Wren6991, 2025-03-31:
It runs fine for me on Firefox on my laptop, but really it needs to be readable on a Pi 3 etc.

I agree this doesn't need to be a single flat page, and we could split the hierarchy across multiple pages to reduce the number of active DOM elements.

Wren6991, 2025-03-31:
Is there a relationship between the displayed TOC depth and keys reachable from the search box?

mudge, 2025-03-31:
Looking at a built `build/jekyll/_data/nav.json` (including the Pico SDK), including only top-level titles and their sections would reduce the list to 359 entries:

* Computers
    * Getting started
        * Getting started with your Raspberry Pi
        * Install an operating system
        * Set up your Raspberry Pi
        * Configuration on first boot
        * Next steps
    * Raspberry Pi OS
        * Introduction
        * Update software
        * Play audio and video
        * Utilities
        * Accessibility options
        * Use Python on a Raspberry Pi
        * Use GPIO from Python
    * Configuration
        * `raspi-config`
        * non-interactive `raspi-config`
        * Displays
        * Audio
        * Networking
        * Screen blanking
        * Users
        * External storage
        * Kernel command line (`cmdline.txt`)
        * Localise your Raspberry Pi
        * Secure your Raspberry Pi
        * Set up a headless Raspberry Pi
        * Host a wireless network from your Raspberry Pi
        * Use a proxy server
        * `boot` folder contents
        * LED warning flash codes
        * Configure UARTs
        * Device Trees, overlays, and parameters
        * Change the default pin configuration
    * `config.txt`
        * What is `config.txt`?
        * `autoboot.txt`
        * Common options
        * Onboard analogue audio (3.5mm jack)
        * HDMI audio
        * Boot Options
        * GPIO control
        * Overclocking options
        * Conditional filters
        * Memory options
        * Licence key and codec options
        * Video options
        * Camera settings
    * Legacy `config.txt` options
        * Legacy options
        * Legacy boot options
        * Legacy GPIO control
        * Legacy overclocking options
        * Legacy conditional filters
        * Legacy memory options
        * Legacy video options
        * Raspberry Pi 4 HDMI pipeline
        * Legacy Miscellaneous Options
    * The Linux kernel
        * Introduction
        * Update
        * Build the kernel
        * Configure the kernel
        * Patch the kernel
        * Kernel headers
        * Contribute
    * Remote access
[... 266 lines cut ...]
        * pico_base <strong class="contexttag RP2350">RP2350</strong>
        * pico_binary_info <strong class="contexttag RP2350">RP2350</strong>
        * pico_bootrom <strong class="contexttag RP2350">RP2350</strong>
        * pico_bit_ops <strong class="contexttag RP2350">RP2350</strong>
        * pico_cxx_options <strong class="contexttag RP2350">RP2350</strong>
        * pico_clib_interface <strong class="contexttag RP2350">RP2350</strong>
        * pico_crt0 <strong class="contexttag RP2350">RP2350</strong>
        * pico_divider <strong class="contexttag RP2350">RP2350</strong>
        * pico_double <strong class="contexttag RP2350">RP2350</strong>
        * pico_float <strong class="contexttag RP2350">RP2350</strong>
        * pico_int64_ops <strong class="contexttag RP2350">RP2350</strong>
        * pico_malloc <strong class="contexttag RP2350">RP2350</strong>
        * pico_mem_ops <strong class="contexttag RP2350">RP2350</strong>
        * pico_platform <strong class="contexttag RP2350">RP2350</strong>
        * pico_printf <strong class="contexttag RP2350">RP2350</strong>
        * pico_runtime <strong class="contexttag RP2350">RP2350</strong>
        * pico_runtime_init <strong class="contexttag RP2350">RP2350</strong>
        * pico_stdio <strong class="contexttag RP2350">RP2350</strong>
        * pico_standard_binary_info <strong class="contexttag RP2350">RP2350</strong>
        * pico_standard_link <strong class="contexttag RP2350">RP2350</strong>
    * External API Headers
        * boot_picobin_headers <strong class="contexttag RP2350">RP2350</strong>
        * boot_picoboot_headers <strong class="contexttag RP2350">RP2350</strong>
        * boot_uf2_headers <strong class="contexttag RP2350">RP2350</strong>
        * pico_usb_reset_interface_headers <strong class="contexttag RP2350">RP2350</strong>

mudge, 2025-03-31:
> Is there a relationship between the displayed TOC depth and keys reachable from the search box?

No, what is available in search is dictated by what Algolia Docsearch crawls out of the actual page content.

mudge, 2025-03-31:
It's worth noting that on sufficiently wide screens, there is a _second_ table of contents specific to the current page shown on the right, e.g.

[image]

kilograham, 2025-03-31:
seems reasonable in the shortened version

mudge, 2025-04-03:
Despite reducing the left-hand table of contents to [26% of its former size](https://github.com/raspberrypi/documentation/pull/4065#issuecomment-2771655300), @JamesH65 (and others via email) are still having issues given the Hardware page still has over 39,000 elements that need their styles recalculating as the page loads.

While I can try to further whittle down the markup in the table of contents, there are over 33,000 elements in the main page content itself.

@lurch made reference to it being a conscious past decision to format these parts of the site as a few, single, very long pages but is this something we'd revisit?

As far as I can see, [postprocess_doxygen_adoc.py](https://github.com/raspberrypi/documentation/blob/fa90756a97eeea6caaa84b3934d94e6efabf753e/scripts/postprocess_doxygen_adoc.py) is actually responsible for splitting the [doxygentoasciidoc](https://github.com/raspberrypi/doxygentoasciidoc) output of a single `all_groups.adoc` into separate `.adoc` files. At the moment, it does this by splitting on the top-level group headings, e.g.

```asciidoc
[#group_hardware,reftext="Hardware APIs"]
```

But perhaps it could be modified to break each inner group out into its own page, e.g. a page for `hardware_adc` and one for `hardware_base`, etc.

For perspective: if we did that, the current group with the largest number of elements is `hardware_adc` with a total of 521 elements.

lurch, 2025-04-03:
To clarify further, I believe Paul is talking about https://www.raspberrypi.com/documentation/pico-sdk/hardware.html

JamesH65, 2025-04-03:
I have always hated (and that is not too strong a word) the move to massively long pages of information, so anything that makes them shorter is OK with me.

JamesH65, 2025-06-08:
ping

kvnloo, 2026-09-29:
Given the left navigation was already reduced substantially and the page still has >30k elements in its content, further ToC markup optimisation looks like diminishing returns.

Splitting the generated Pico SDK pages by inner Doxygen group seems like the stronger next boundary:

`hardware.html` → `hardware_adc.html`, `hardware_dma.html`, `hardware_flash.html`, etc.

Search should remain unaffected since Algolia crawls page content independently of the navigation tree, and the right-side per-page ToC can retain detailed function navigation.

I'd measure this with a simple performance budget on a lower-end Pi as well as desktop:

- DOM node count
- style/layout time
- initial page load
- navigation latency

That turns the restructuring into a measurable performance improvement rather than only an information-architecture change.

stevedavies-rp, 2026-11-09:
We're now developing a new toolchain that will allow us more freedom and flexibility with our TOC and high-level IA. Marking this as closed for now, but expect to see changes start to come through in 2027.

stevedavies-rp, 2026-11-09:
Tracked internally.

closed 2026-11-09 by stevedavies-rp