raspberrypi/documentation #4097: The pages that refer to Touchdisplay 2 probably need rearchitecting

JamesH65, 2025-05-09:
I've added some CM specific data in another PR, but the pages could really do with a reshuffle to make them a bit saner. There is duplicate data and its clear the TD2 stuff was added later when it came out.

documentation/asciidoc/accessories/touch-display-2/about.adoc
documentation/asciidoc/accessories/display/display_intro.adoc
documentation/asciidoc/computers/compute-module/cmio-display.adoc

lurch, 2025-05-09:
(I preemptively added this to "backlog", as I assume that no-one has time to tackle this any time soon, and we don't want the stalebot to auto-close this issue)

lurch, 2025-05-18:
#4002 is also asking for extra clarification that when using a Touchdisplay with the Pi 5, you need to use the specific [15->22 pin DSI cable](https://www.raspberrypi.com/products/display-cable/), you can't use a [15->22pin CSI cable](https://www.raspberrypi.com/products/camera-cable/).