raspberrypi/documentation #4132: CM5 IO Board instructions need jumpers for CAM1 access

obugandi, 2025-06-28:
The instructions say that the board needs J6 jumpers to enable CAM0 just like the CM4, when the CM5's board has written that it needs jumpers for enabling CAM0.

https://www.raspberrypi.com/documentation/computers/compute-module.html#attach-a-camera-module 

See the text "(CM4 and CM5): Connect the J6 GPIO pins with two vertical-orientation jumpers."  in the section for connecting the 2nd camera (on CAM0). The board's image partially shows the text "and DISP0" which are the instructions for jumping in order to enable CAM0/DISP0.

In this image https://github.com/raspberrypi/documentation/pull/4131/files#diff-6f5e64216c15d24d03c5ca395cde0ddebcce88dcecefcabd983a688d59908526 you can see the CM5's board says to fit the jumpers for CAM/DISP1

obugandi, 2025-06-28:
I filed this bug after submitting a PR to address this issue.

PR here https://github.com/raspberrypi/documentation/pull/4131

katshann-raspberry, 2025-09-25:
Associated PR is now merged and sent to publish. Closing as completed. 
Cheers!

closed 2025-09-25 by katshann-raspberry