raspberrypi/documentation #4147: Software Camera Sync options not documented

gnbl, 2025-07-28:
I was excited to read

https://www.raspberrypi.com/documentation/computers/camera_software.html#software-camera-synchronisation

and bought a couple of Pi Zero 2 Ws and cameras.

The example given in the documentation seems to assume 2 local cameras on a Pi 5, but not to worry, settings are configurable:

> The server’s broadcast address and port, the frequency of the timing messages and the number of frames to wait for clients to synchronise, can all be changed in the camera tuning file. Clients only pay attention to the broadcast address here which should match the server’s; the other information will be ignored. Please refer to the [Raspberry Pi Camera tuning guide](https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf) for more information.

(Source file: https://github.com/raspberrypi/documentation/blob/021c6201d4e64aad91896e7dadccb3c26aeadd92/documentation/asciidoc/computers/camera/rpicam_apps_multicam.adoc?plain=1#L63)

However, I found no mention of software sync options in the PDF, and the tuning files (old and new) for the Camera 3 contain only an empty object with index "rpi.sync".

Please let me know where to find this information.

davidplowman, 2025-07-29:
It turns out the Tuning Guide had been updated, but not published. Documents now seem to go into some PIP portal, and the old datasheets location is deprecated and not to be updated. Trying to figure out how to get the thing into Cognidox and thence to PIP. So straightforward.

davidplowman, 2025-07-29:
Apparently you can find the latest tuning guide here:
https://pip.raspberrypi.com/categories/1259-audio-camera-and-display/documents/RP-009070-TC/Camera-Algorithm-and-Tuning-Guide.pdf

gnbl, 2025-07-29:
Thanks!
However, it seems unless you have specific network requirements, the default broadcast/multicast IP4 address "just works" for server and client alike, so I did not actually have to change anything from the example except (remove) the camera index... 😅

gnbl, 2025-07-30:
On second thought, maybe add a sentence to the online docs, that the broadcast address does not have to be set to the devices' IP?

lurch, 2025-07-30:
ping @njhollinghurst

closed 2025-10-19 by github-actions[bot]